raspberrypi/pico-setup-windows #8: Unable to build after install on new project

mytechnotalent, 2023-03-04:
After installing `pico-setup-windows-x64-standalone` when I load `Pico - Visual Studio Code` and created a new project, when I try to CMAKE:PROJECT OUTLINE and build a copy of the `picow_tcp_client.c` I get the below.  All files are in the root of the project.

```
CMakeLists.txt
lwipopts_examples_common.h
lwipopts.h
pico_sdk_import.cmake
picow_tcp_client.c
```

```
picow_tcp_client.c:11:10: fatal error: pico/cyw43_arch.h: No such file or directory
[build]    11 | #include "pico/cyw43_arch.h"
[build]       |          ^~~~~~~~~~~~~~~~~~~
[build] compilation terminated.
```

my `CMakeLists.txt`  

```

cmake_minimum_required(VERSION 3.13)

# Pull in SDK (must be before project)
include(pico_sdk_import.cmake)

project(badge1 C CXX ASM)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

set(PICO_EXAMPLES_PATH ${PROJECT_SOURCE_DIR})

# Initialize the SDK
pico_sdk_init()

add_compile_options(-Wall
        -Wno-format          # int != int32_t as far as the compiler is concerned because gcc has int32_t as long int
        -Wno-unused-function # we have some for the docs that aren't called
        -Wno-maybe-uninitialized
        )

if (PICO_CYW43_SUPPORTED) # set by PICO_BOARD=pico_w
    if (NOT TARGET pico_cyw43_arch)
        message("Skipping Pico W examples as support is not available")
    else()

        if (DEFINED ENV{WIFI_SSID} AND (NOT WIFI_SSID))
            set(WIFI_SSID $ENV{WIFI_SSID})
            message("Using WIFI_SSID from environment ('${WIFI_SSID}')")
        endif()

        if (DEFINED ENV{WIFI_PASSWORD} AND (NOT WIFI_PASSWORD))
            set(WIFI_PASSWORD $ENV{WIFI_PASSWORD})
            message("Using WIFI_PASSWORD from environment")
        endif()

        set(WIFI_SSID "${WIFI_SSID}" CACHE INTERNAL "WiFi SSID for examples")
        set(WIFI_PASSWORD "${WIFI_PASSWORD}" CACHE INTERNAL "WiFi password for examples")
    endif()
endif()

set(WIFI_SSID "${WIFI_SSID}" CACHE INTERNAL "WiFi SSID for examples")
set(WIFI_PASSWORD "${WIFI_PASSWORD}" CACHE INTERNAL "WiFi password for examples")

add_executable(badge1
        picow_tcp_client.c
        )
target_compile_definitions(badge1 PRIVATE
        WIFI_SSID=\"${WIFI_SSID}\"
        WIFI_PASSWORD=\"${WIFI_PASSWORD}\"
        TEST_TCP_SERVER_IP=\"${TEST_TCP_SERVER_IP}\"
        )
target_include_directories(badge1 PRIVATE
        ${CMAKE_CURRENT_LIST_DIR}
        ${CMAKE_CURRENT_LIST_DIR}/.. # for our common lwipopts
        )
target_link_libraries(badge1
        pico_cyw43_arch_lwip_threadsafe_background
        pico_cyw43_arch_lwip_poll
        pico_stdlib
        )

pico_add_extra_outputs(badge1)
```

mytechnotalent, 2023-03-04:
Update

Setting the board type explicitly within `CMakeLists.txt` fixes this.  I will close this issue but wanted to share with the dev team so they may think about adding this info to the readme.  

```
# Set the board type
set(PICO_BOARD pico_w)
```

closed 2023-03-04 by mytechnotalent

ndabas, 2023-03-05:
Thanks for figuring it out and documenting your fix.

I'd like to point out that this is not specific to the Windows installer, it is about using the SDK and the specific example you chose. Setting `PICO_BOARD` is [documented in the pico-examples ReadMe](https://github.com/raspberrypi/pico-examples#pico-w-networking): (where you copied picow_tcp_client.c from)
> These examples are for the Pico W, and are only available for PICO_BOARD=pico_w

The Windows + VS Code specific instructions for setting `PICO_BOARD` are [in the FAQ](https://github.com/raspberrypi/pico-setup-windows/wiki/FAQ).