raspberrypi/pico-setup-windows #9: Windows create new project tutorial command does not include vscode settings

JazzBrown1, 2023-03-14:
Hi

I followed the instructions on this page to create a new project and realised the .vscode folder that was copying was empty.

(https://github.com/raspberrypi/pico-setup-windows/tree/master/docs)/tutorial.md

I assume the contents should be copied too? If so I think this is the correct command: 

Copy-Item ${env:PICO_EXAMPLES_PATH}\.vscode -recurse

ndabas, 2023-03-17:
Correct, that command needs the `-Recurse` parameter added. Will be fixed in the next release.

bzuidgeest, 2023-05-28:
To be a little pedantic. I noticed that when I tried the fixed command from @JazzBrown1 is missing a "\"
should be Copy-Item ${env:PICO_EXAMPLES_PATH}**\**.vscode -recurse

ndabas, 2023-06-14:
This is fixed in the [latest release](https://github.com/raspberrypi/pico-setup-windows/releases/latest).