raspberrypi/pico-setup-windows #10: Windows 11 Not supported (possible on an ARM install only)

dcomer, 2023-03-16:
Clearly not all users are using Parallels on an Apple Silicon system, but it is very unfortunate that, despite having Rosetta, that one cannot use this project to install in this environment. I, fortunately have a x86 Mac as well and will try on that platform.

This is not to knock this project. I greatly appreciate the work. Just wish there was a Mac Silicon fork.

Thank you.

aallan, 2023-03-16:
The installer works under Windows 11 on an x86 system, so this is an Arm-only problem. 

Although, that said. If you have an Arm-based Mac, why are you attempting to install the toolchain under Windows running in Parallels? Why not just install it under macOS? 

The instructions for installation on Mac (both Arm and x86) are documented in section 9.1 of the "[Getting Started](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)" guide and work out the box (I write the documentation, and my daily driver is a Mac, the Mac instructions are always going to work out of the box).

dcomer, 2023-03-16:
Thanks for your reply.

I agree it is an Arm-only problem. It would still be a macOS problem (remember, this is all on an Apple Silicon machine (M2 in this case). But, that is a minor detail. The issue isn't the toolchain. Arm's Apple Silicon release works fine. The problem (for me) is understanding the various config files that VS Code requires. There are a number of YouTube videos that come close to a good explanation but not that completely (that I've seen) thoroughly explain the VS Code configuration files.

This project looked like a good opportunity to see a functioning install to understand VS Code. I could then use the knowledge gained and port over what I needed with some modifications.

dcomer, 2023-03-16:
I am closing the issue as this is probably a tangent for the project, and a project dedicated for MacOS users would be a nice idea.

closed 2023-03-16 by dcomer

aallan, 2023-03-17:
>  …and a project dedicated for MacOS users would be a nice idea.

I'm am currently looking about for someone to work on a macOS installer. It'd probably be not much more than something like the [pico-setup script](https://github.com/raspberrypi/pico-setup/) for Raspberry Pi OS wrapped inside a PKG installer file.

ndabas, 2023-03-17:
If you're just looking for the VS Code config, the [config files in this project](https://github.com/raspberrypi/pico-setup-windows/tree/master/packages/pico-examples/ide/vscode) *should* work just fine. Just copy them into a `.vscode` directory in your project. The files assume that you have the toolchain installed, `PICO_SDK_PATH` configured, etc. which the pico-setup script linked to by @aallan does for you.

As for a complete macOS installer, I guess that would be the setup script, the VS Code config, and possibly pre-compiled versions of OpenOCD, picotool, pioasm, and elf2uf2.

dcomer, 2023-03-17:
@ndabas Good suggestion. I will take a look at your suggestion and see how far it takes me. 
@aalan I have never written a dmg/pkg for the mac but have found a decent tutorial on how to do so. I will take a look and see if I am able to make progress with doing so. It makes take me a week or two (due to prior commitments). This is definitely needed for the macros community. Given that the ARM GNU toolchain is available for both the x86 and the Apple silicon, my approach would be to install the toolchain in the ~/<user>/opt or /opt (all users) directory and make that an option. I have no issues with a brew or other solution install, but I prefer a purist approach as I come from both a Windows developer and Linux Developer background. I will get back to you in the next week or two to let you know my progress. It's possible someone may beat me to it, but that's OK. The goal is to create something that mac users can benefit from.

ghollingworth, 2023-03-17:
Note, we are looking at a different solution to this for supporting the toolchain on MacOS.  So if you do this, it may end up not being used, or at least 

Basically, we'd prefer to install the toolchain through a VSCode extension, which would then make it possible to support Linux, MacOS and Windows all through the same mechanism...  But we're still looking into it and don't know whether this will actually be the right solution...

dcomer, 2023-03-17:
Understood. What exactly does installing through a VSCode extension entail? I was under the impression that the toolchain was installed in a location and VSCode references the toolchain by an environment variable? Thanks.

dcomer, 2023-03-17:
Got it. Thanks. I will reconsider how I had planed to proceed (or not).