raspberrypi/pico-setup-windows #11: After first installation cmake is not working anymore

AndreasMainz, 2023-04-04:
With the first setup everything works fine, eben Debugger is working, but if you leave
VSC and restart, even the include path for the stdlib.h is lost.
Cmake reports:
Bad CMake executable: "". Check to make sure it is installed or the value of the "cmake.cmakePath" setting contains the correct path.

Forgot to write some vars permantently?

ndabas, 2023-04-04:
This is by design. As explained in the [ReadMe](https://github.com/raspberrypi/pico-setup-windows/blob/master/docs/tutorial.md) that's included with the installer, you need to use the _Pico - Visual Studio Code_ shortcut to launch VS Code with the correct environment variables.

closed 2023-04-04 by ndabas

AndreasMainz, 2023-04-04:
many thanks..now it works..
i have also problems with vsc in ubuntu. do i need also to run a special script for ubuntu?
Many thanks for your response.

AndreasMainz, 2023-04-04:
Hi Nikhil,

what do i have to consider, if i want to run Code under Ubuntu. Also here a lot
of enviroment vars are not set after Installation with script:
pico_setup.sh
Is there a simular trick?

Many thanks for your Reply and best regards
Andreas

Von: Nikhil Dabas
Gesendet: Dienstag, 4. April 2023 21:26
An: raspberrypi/pico-setup-windows
Cc: Andreas Mainz; Author
Betreff: Re: [raspberrypi/pico-setup-windows] After first installation cmakeis not working anymore (Issue #11)

This is by design. As explained in the ReadMe that's included with the installer, you need to use the Pico - Visual Studio Code shortcut to launch VS Code with the correct environment variables.

ndabas, 2023-04-04:
If you have used the pico_setup.sh script from the [pico-setup](https://github.com/raspberrypi/pico-setup) repo, it adds the required environment variables to `~/.bashrc`. You could try manually dot-sourcing this file if you don't have the environment setup automatically.

You can also ask in the [forum](https://forums.raspberrypi.com/viewforum.php?f=144) if this doesn't resolve your issue on Ubuntu.