raspberrypi/pico-setup-windows #12: python installed without tkinter?

ullibak, 2023-04-06:
I installed the SDK by running "pico-setup-windows-x64-standalone.exe". Everything was installed without error messages. Example projects compile without problems.

Then, I tried to run the Pico Project Generator (github.com/raspberrypi/pico-project-generator) with the "--gui" option and found that tkinter was missing. 
A search on the internet revealed that with newer versions of Python, tkinter ist always included. Here, version 3.9.13 installed, but tkinter is missing. 
Is there an issue in the Python installation routine or how can I install tkinter manually?

Thx.

lurch, 2023-04-22:
According to https://github.com/raspberrypi/pico-setup-windows/blob/master/config/x64-standalone.json it installs `python-3.9.13-embed-amd64.zip` and https://docs.python.org/3/using/windows.html#windows-embeddable says "Tcl/tk (including all dependants, such as Idle), pip and the Python documentation are not included."

ping @ndabas (author of this repo) and @JamesH65 (author of [pico-project-generator](https://github.com/raspberrypi/pico-project-generator) )

ullibak, 2023-04-23:
Thanks for the clarification. So this is a feature not a bug :-). 
For users of the python project generator GUI the simplest solution is most probably to (additionally) install a regular Python package and change the path to python.exe in "....\Pico SDK v1.5.0\pico-env.cmd". At least, this works fine for me.

closed 2023-04-23 by ullibak

ndabas, 2023-04-23:
Yes, that's the thing -- we can only include the embedded distribution in our installer, which doesn't have support for Tcl/tk. The Python included with the SDK installer is sufficient only for the SDK build process, and isn't intended for general-purpose use beyond that.

The workaround, like you've figured out, is simply to install Python as usual. You do not need to edit the path if you launch the Pico project generator from outside the Pico developer prompts or VS Code.

lurch, 2023-04-23:
> You do not need to edit the path if you launch the Pico project generator from outside the Pico developer prompts 

But I guess if you do that, then the "Build after project created" option of pico-project-generator won't work?

ullibak, 2023-04-23:
With the modifications in pico-env.cmd I can use the pico command prompt environment that contains all the paths etc. and have a "full" python at my disposal.
If I use an "ordinary" command prompt, the project generator does not even start. Error message is "Unable to find the 'arm-none-eabi-gcc' compiler....".
So the GUI-version of the project generator needs a "pico environment" and tkinter at the same time.

ndabas, 2023-04-23:
I guess the solution might be to create a Windows installer for pico-project-generator that fixes everything up properly and adds a shortcut to the Start menu to launch it. Modifying the path in a batch file, while a workaround for now, is not something that we should expect all users to do.

JamesH65, 2023-04-24:
Hmm, a bit of a PITA to fix I suspect. I'd hoped that the Project generator would be installed just by cloning the repo or copying the .py file, along with associated installs of Python/TkInter on all platforms. Just set the SDK path and it should all hang together. Sadly this appears not to be the case on Windows. You can set the compiler path on the command line with --cpath which might be an option, in combination with the SDK path. Or maybe a small install batch file for Windows although I am a bit out of touch with Windows stuff nowadays to know if that would be sufficient.

lurch, 2023-04-24:
> The 'proper' way to do it would be to somehow get a Python distribution with Tcl/tk support in the SDK installer.

Just out of curiosity, is there a reason we can't do that? https://docs.python.org/3/using/windows.html#installing-without-ui

ndabas, 2023-04-25:
> Just out of curiosity, is there a reason we can't do that? https://docs.python.org/3/using/windows.html#installing-without-ui

Yes. One of the goals of this Windows installer is to have a stable distribution which will continue to work exactly as it does today. If we run the Python installer instead of bundling the embedded distribution, it will not actually install the version we want if the user already has a newer version installed, which can potentially break stability for us.

This is actually the approach that I had before it became the official installer. There is a [configuration file](https://github.com/raspberrypi/pico-setup-windows/tree/master/config) - x64.json - which will build the installer that way. However that was always problematic because installing anything globally can break for many reasons.

ndabas, 2023-04-25:
Okay there is a hack-free way to use the Pico project generator on Windows:
1. Install the SDK with this installer.
2. Install the full version of Python from https://www.python.org/downloads/windows/.
3. Open a _Pico - Developer Command Prompt_ window.
4. `cd` to the location where you have downloaded/cloned pico-project-generator.
5. Launch using py.exe instead of python.exe: `py pico_project.py --gui` -- this will use the latest installed version (the full version) instead of the embedded distribution with the SDK.

JamesH65, 2023-04-25:
> Okay there is a hack-free way to use the Pico project generator on Windows:
>
> 1. Install the SDK with this installer.
> 2. Install the full version of Python from [...]

Thanks for figuring that out. I'll update the README with this information.

ullibak, 2023-04-26:
Tested in windows sandbox. Works fine. Thanks!
If you add a .cmd file (e.g. GenerateShortcut.cmd) to the project generator archive containing

```
echo cd %cd% > "%PICO_INSTALL_PATH%\generator.cmd"
echo py.exe pico_project.py --gui >> "%PICO_INSTALL_PATH%\generator.cmd"
```
the user can simply start a Pico Developer Comand Prompt, cd to the location of the project generator and run this cmd file once. From then on, it is sufficient to open the Developer Command Prompt and simply type "generator" to start the project generator.