raspberrypi/pico-setup-windows #15: Compiler 'arm-none-eabi-gcc' not found

philwu, 2023-04-27:


ghollingworth, 2023-04-27:
It looks like it’s using the Microsoft visual studio version of cmake…

philwu, 2023-04-27:
> 

That's right. I use cmake from pico sdk and problem is gone. Thank you very much.

closed 2023-04-27 by philwu

Vodlex1, 2023-07-08:
> > 
> 
> That's right. I use cmake from pico sdk and problem is gone. Thank you very much.

Hello, I'm facing the exact same issue and I just wanted to ask how you switched over to the pico sdk cmake?
Thanks.

ndabas, 2023-07-08:
> Hello, I'm facing the exact same issue and I just wanted to ask how you switched over to the pico sdk cmake?

Just make sure you launch VS Code or the command prompt using the "Pico - ..." shortcuts.

Mola19, 2024-09-11:
No idea if i overread that add some point or not, but apparently this issue appears if you don't restart after installing.