raspberrypi/pico-setup-windows #16: Start menu links not all being installed

willeccles, 2023-05-05:
After installing the PICO SDK, I only get the following start menu links:

[image]

This is on my work machine, so it was not a surprise that my already complicated environment would cause issues. However, I had a coworker with no software installed (he's a mechanical guy) run the installer, and he had the same results. We're both on Windows 10. The installer log shows that all was well, and it claims that it installed all the links, but we don't see them:

[image]
[image]

Directly running `pico-code.ps1` works fine for both of us, so everything appears to have installed correctly except for the shortcuts themselves. I do most of my development work on Linux, so this is the first time I have used the Windows installer. I'm not sure if I've missed something. I have tried rerunning the installer which simply uninstalls and re-installs with the same results.

I suspect the fact that the first shortcut was successfully created and the second and third were not might be a hint at the root cause, but I am not familiar enough with Windows installer packages to know for sure.

ndabas, 2023-05-06:
That's interesting, I'll investigate.

Since you mentioned it's your work machine, could you give me some more info on the environment?
- What is the exact version and edition of Windows 10 that you have? Enterprise?
- Is there some corporate-mandated anti-virus running there? My guess is that the two shortcuts that have disappeared are both shortcuts to PowerShell scripts, so some anti-malware software might consider that a risk somehow, and deleted those immediately on creation. It would be great if you are able to check the logs (or list of removed/quarantined files) of any such software as well, if you have access to do so.

willeccles, 2023-05-06:
Thanks for looking into this. I will investigate further when I'm back in the office on Monday. I want to say we have Pro or Enterprise. We do have an antivirus; I'll see if I can check if it had anything to do with this.

willeccles, 2023-05-08:
Alright, here's the Windows version:
[image]

I'll get back to you on the AV situation, I have to figure out where I'd look to find such a thing.

willeccles, 2023-05-08:
Okay, so we have Sentinel One. Looks like it has reported no issues or quarantined files in the last week, so it's not that. Are there any other policies or anything that might prevent installing such a file?

In the meantime, is there anything you can recommend to fix this problem? I would like to be able to write the developer documentation for this project to say nothing more than "follow the instructions for installation and then open this program." Unfortunately, it seems a little more complicated than that at the moment.

ndabas, 2023-05-10:
I'll look into this. I haven't been able to replicate the issue in several test runs so far, but I'll try to get the exact Windows version you have and test with that (it's relatively old for a "Pro" build -- most business who need to stick to a stable/LTS version use Enterprise or Education.)

lurch, 2023-05-10:
If it's a work machine, perhaps your IT support team might be able to give you some hints? :shrug:

willeccles, 2023-05-11:
As always, I was trying to avoid having to talk to IT, but alas. I'll have to shoot them an email today when I've got a few minutes.

y75yan, 2023-08-08:
I had the identical issue, running W11 Edu and Sentinel One. The latter blocked the installation. 
[image]
[image]

ndabas, 2023-08-08:
The fix is identical as well -- speak to your IT staff or contact the anti-malware developer. It is obviously a false positive since our software does not do anything malicious.

ndabas, 2024-01-25:
Closing because this isn't an issue we can fix here.

closed 2024-01-25 by ndabas