raspberrypi/pico-setup-windows #17: Installation from normal (not Administrator) account

SteveCan1, 2023-05-13:
Hi,
When working in Windows, I generally like to work in a regular account, but when trying to install the Pico-SDK environment in Win11, I encountered errors and was unable to complete the installation.

I believe this stems from the choice of directories in which the SDK and sample files are stored...

Is it possible to override this location, so the files themselves can be stored in a "public" directory structure on the machine?

Thanks,
Steve

ndabas, 2023-05-15:
Could you elaborate on the specific errors that you encountered?

The installer needs administrative permission to install and it installs into the standard recommended location by default (Program Files). When you run the installer it does give you an option to choose the directory to install to.

SteveCan1, 2023-05-16:
Hi,
Thanks for responding to my issue...
When I first ran the installer I did not notice that the directory proposed for installation was a directory located in the Administrator's account... Apparently when the second portion of the script is run it no longer had administrator rights(?) and was unable to write into those directories.

Changing the selected directory to use a publicly-assessible location worked fine - everything is installed and seems to work fine, though I do have one additional question: There is no simple "Build" option to compile the code - I have "Start Debugging" as well as "Run Without Debugging" options. How can I simply compile the code (to generate the .uf2 file) without getting the following error:
[image]

Thanks again for your assistance and pointing out my oversight. This issue can be closed (I will look through the forums to see if anyone has addressed the build issue...

Best,
Steve

closed 2023-05-16 by SteveCan1