raspberrypi/pico-setup-windows #18: Setup through proxy

SteveCan1, 2023-05-13:
Hi,

Is it possible to configure the setup executable through a Windows proxy?

Running the pico-windows-setup program from a location requiring a proxy fails, and supporting a proxy could simply setup of the overall environment...

Thanks,

Steve

ndabas, 2023-05-15:
Could you elaborate on what exactly fails during the installation? The only thing that's downloaded at install time is Visual Studio Code (because they have frequent updates) and the installer is designed to ignore VS Code download/install failures. If that is the issue, you can just install VS Code yourself manually.

SteveCan1, 2023-05-16:
Hi Nikhil,
Thanks for responding to my question.

In fact, the error seems to be limited to the pico-extras cloning/compiling operation.

The screen shows the following:

[image]

Cloning into 'pico-extras'...

fatal: unable to access 'https://github.com/raspberrypi/pico-extras.git&#39;: Failed to connect to github port 443 after 21126 ms: Couldn't connect to server

Everything else seems to have installed fine (including Visual Studio Code), but since the pico-extras did not get properly cloned, the initial build of all the binaries did not occur.

I was able to build them using VSCode and everything seems fine.

Thanks for your help.

Regards,

Steve

ndabas, 2023-06-14:
This should be fixed in the [latest release](https://github.com/raspberrypi/pico-setup-windows/releases/latest). The pico-extras and pico-playground repos are now included in the installer instead of being downloaded later.

closed 2023-06-14 by ndabas