raspberrypi/pico-setup-windows #19: Missing components

mohdrais, 2023-05-14:
How to overcome the missing components as shpwn in the Pico-Developer Command Pr
[image]
ompt

ndabas, 2023-05-15:
This should really not happen after a successful install -- are all of the tools present and installed into `C:\Program Files\Raspberry Pi\Pico SDK v1.5.0`? You should have folders like `cmake`, `python`, `gcc-arm-none-eabi`, and so on for all of the tools.

Additionally, I see the screenshot is from Windows Terminal, does starting the Pico - Developer Command Prompt from the Start menu work?

mohdrais, 2023-05-18:
Yes, all the folders mentioned (eg cmake, python etc) are installed. Pico-Developer Command Prompt does  work as shown in the picture previously
[image]
but Pico- DeveloperShell does not work

closed 2024-08-15 by ghollingworth