raspberrypi/pico-setup-windows #23: No way to build Pico W examples from VSCode Cmake tab

mdrobot7, 2023-06-19:
Basically what the title says -- building for Pico W require extra build flags specifying that the board is a Pico W, network SSID, and network password. The Cmake tab doesn't pick up the Pico W examples as Cmake projects, and there is no way to set these flags to build them.

Setup script version: 1.5.1
OS: Windows 11 Pro

mdrobot7, 2023-06-21:
Yeah, I didn't see that. I followed the instructions on the wiki to set the board and build variables, and the build failed halfway through, saying it couldn't find a dependency in the SDK. Build output is below. I don't really know what's going on

I built the pico w examples on another machine successfully through the command line (after using the setup script, needing to build for pico w, and not knowing how to set the build flags through GUI). I followed the instructions in section 2.2 [here](https://datasheets.raspberrypi.com/picow/connecting-to-the-internet-with-pico-w.pdf) to configure cmake, then hit Build All in the Cmake tab and it built successfully. I don't know how the VSCode implementation differs, maybe it's just not grabbing a flag.

Section of build output:
```
[build] [5465/13794  39% :: 232.816] Building C object pico_w/bt/standalone/CMakeFiles/picow_ble_temp_sensor_with_wifi.dir/C_/Users/Michael/DATA/Projects/Raspberry_Pi/Pico_SDK_v1.5.1/pico-sdk/src/rp2_common/pico_cyw43_driver/cybt_shared_bus/cybt_shared_bus.c.obj
[build] FAILED: pico_w/bt/standalone/CMakeFiles/picow_ble_temp_sensor_with_wifi.dir/C_/Users/Michael/DATA/Projects/Raspberry_Pi/Pico_SDK_v1.5.1/pico-sdk/src/rp2_common/pico_cyw43_driver/cybt_shared_bus/cybt_shared_bus.c.obj 
[build] C:\Users\Michael\DATA\Projects\RASPBE~1\PICOSD~1.1\GCC-AR~1\bin\AR19DD~1.EXE -DCYW43_ENABLE_BLUETOOTH=1 -DCYW43_LWIP=1 -DENABLE_BLE=1 -DLIB_PICO_ASYNC_CONTEXT_THREADSAFE_BACKGROUND=1 -DLIB_PICO_BIT_OPS=1 -DLIB_PICO_BIT_OPS_PICO=1 -DLIB_PICO_BTSTACK_BLE=1 -DLIB_PICO_BTSTACK_CYW43=1 -DLIB_PICO_BTSTACK_FLASH_BANK=1 -DLIB_PICO_CYW43_ARCH=1 -DLIB_PICO_DIVIDER=1 -DLIB_PICO_DIVIDER_HARDWARE=1 -DLIB_PICO_DOUBLE=1 -DLIB_PICO_DOUBLE_PICO=1 -DLIB_PICO_FLASH=1 -DLIB_PICO_FLOAT=1 -DLIB_PICO_FLOAT_PICO=1 -DLIB_PICO_INT64_OPS=1 -DLIB_PICO_INT64_OPS_PICO=1 -DLIB_PICO_MALLOC=1 -DLIB_PICO_MEM_OPS=1 -DLIB_PICO_MEM_OPS_PICO=1 -DLIB_PICO_PLATFORM=1 -DLIB_PICO_PRINTF=1 -DLIB_PICO_PRINTF_PICO=1 -DLIB_PICO_RAND=1 -DLIB_PICO_RUNTIME=1 -DLIB_PICO_STANDARD_LINK=1 -DLIB_PICO_STDIO=1 -DLIB_PICO_STDIO_UART=1 -DLIB_PICO_STDLIB=1 -DLIB_PICO_SYNC=1 -DLIB_PICO_SYNC_CRITICAL_SECTION=1 -DLIB_PICO_SYNC_MUTEX=1 -DLIB_PICO_SYNC_SEM=1 -DLIB_PICO_TIME=1 -DLIB_PICO_UNIQUE_ID=1 -DLIB_PICO_UTIL=1 -DPICO_BOARD=\"pico_w\" -DPICO_BUILD=1 -DPICO_CMAKE_BUILD_TYPE=\"Release\" -DPICO_COPY_TO_RAM=0 -DPICO_CXX_ENABLE_EXCEPTIONS=0 -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DPICO_EXTRAS=1 -DPICO_NO_FLASH=0 -DPICO_NO_HARDWARE=0 -DPICO_ON_DEVICE=1 -DPICO_PROGRAM_URL=\"https://github.com/raspberrypi/pico-examples/tree/HEAD/pico_w/bt/standalone\" -DPICO_TARGET_NAME=\"picow_ble_temp_sensor_with_wifi\" -DPICO_USE_BLOCKED_RAM=0 -DWIFI_PASSWORD=\"password\" -DWIFI_SSID=\"network\" -IC:/Users/Michael/DATA/Projects/pico-examples/pico_w/bt/standalone -IC:/Users/Michael/DATA/Projects/pico-examples/build/pico_w/bt/standalone/generated -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/common/pico_stdlib/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_gpio/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/common/pico_base/include" -IC:/Users/Michael/DATA/Projects/pico-examples/build/generated/pico_base -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/boards/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/pico_platform/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2040/hardware_regs/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_base/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2040/hardware_structs/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_claim/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_sync/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_irq/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/common/pico_sync/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/common/pico_time/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_timer/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/common/pico_util/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_uart/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_resets/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_clocks/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_pll/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_vreg/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_watchdog/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_xosc/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_divider/include" -I"C:/Users/Michael/DATA/Projects/Raspberry Pi/Pico SDK v1.5.1/pico [... cut ...]
      "visibility": "hidden"
    },
    "build": {
      "visibility": "hidden"
    },
    "buildTarget": {
      "visibility": "hidden"
    }
  },
  "cmake.buildBeforeRun": true,
  "cmake.configureOnOpen": true,
  "cmake.configureSettings": {
    "CMAKE_MODULE_PATH": "${env:PICO_INSTALL_PATH}/pico-sdk-tools",
    "PICO_BOARD": "pico_w",
    "WIFI_SSID": "ssid",
    "WIFI_PASSWORD": "password"
  },
  "cmake.generator": "Ninja",
  "C_Cpp.default.configurationProvider": "ms-vscode.cmake-tools"
}
```

ndabas, 2023-06-21:
I believe what's happening here is that the build output paths are too long for CMake/Ninja to handle on Windows. Windows itself supports longer paths but Ninja and CMake have not yet integrated that functionality. The workaround is to set the build output directory to a shorter path, as explained in this FAQ entry:
https://github.com/raspberrypi/pico-setup-windows/wiki/FAQ#cmake-warns-the-object-file-directory--has-n-characters-the-maximum-full-path-to-an-object-file-is-250-characters-see-cmake_object_path_max-object-file--cannot-be-safely-placed-under-this-directory-the-build-may-not-work-correctly

You could also use the `subst` command on Windows to map a drive letter to your project directory, which will shorten the paths too. There is an example of how to do this in the test script here:
https://github.com/raspberrypi/pico-setup-windows/blob/acbad2b5924e4656460a261cb1436dbdad3ae34e/tests/pico-build.cmd#L11

In any case, make sure you reconfigure with CMake after making these changes.

mdrobot7, 2023-06-22:
Ok, thanks

closed 2023-06-22 by mdrobot7

ndabas, 2023-06-22:
Was that the issue? Did the suggestions help fix it?