raspberrypi/pico-setup-windows #24: New install not working

ParrSt, 2023-07-03:
I tried to remove all the old components because I constantly had issues I could not figure out.  I think I may not have had all the componentes linked properly.  I was quite excited to find a full install for windows but when I tried to compile anything I get this errror

CMake Error at CMakeLists.txt:7 (project):Running
 'nmake' '-?'
failed with:
 The system cannot find the file specifiedCMake (project)

As near as I can find that is one of the most key commands for the CMakeLists.txt.  Any ideas?

Bob

ndabas, 2023-07-05:
Could you provide some more info on what exactly you've tried? Are you trying to build from the command line or from Visual Studio Code? Did you start the command prompt or VS Code from the Pico-specific shortcuts? Are you building your own project or one of the examples?

One of the issues I see is that CMake is trying to invoke `nmake`, which should not happen if the environment is set up properly -- in which case Ninja is used as the default build tool instead of `nmake`.

ParrSt, 2023-07-06:
I am using VS Code, both from the shortcut that was installed and by calling 'code' from the developer prompt.  I moved Blinky from the examples location and tried to compile from another location.  (the installer did not give me a choice for location when I installed the examples and I keep my data on a drive other than C:).  I used the following CMakeList...

`cmake_minimum_required(VERSION 3.13)
include(pico_sdk_import.cmake)
project(blinky C CXX ASM)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
pico_sdk_init()
add_executable(blinky blinky.c)
target_link_libraries(blinky pico_stdlib)
pico_add_extra_outputs(blinky)`

Also it doesn't choose the compiler by default, I have to select it from the list.  I choose (GCC 10.3.1 arm-none-eabi) and get a message about unable to find it on my D drive and using the version that is on the C drive, but I don't see an option to select a version on the C drive.
I did try to uninstall and delete the old environment but it looks like something stayed behind (probably registry).  I think you are right that VS Code is trying to use nmake.  Where would I change that?  Or should I uninstall and try to clean out the registry then re-install?

Bob

lurch, 2023-07-06:
> I moved Blinky from the examples location and tried to compile from another location.

You can't (easily) do that - the example files in pico-examples are only setup to be built from inside the pico-examples repo. For building standalone projects, see Chapter 8 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

ParrSt, 2023-07-06:
> You can't (easily) do that - the example files in pico-examples are only setup to be built from inside the pico-examples repo. For building standalone projects, see Chapter 8 of [...]

This was the easiest and smallest example.  I have the same issue with anything.  I am pretty sure ndabas was correct about the configuration.

ParrSt, 2023-07-06:
Okay, just to be sure I did try following the instructions from Chapter 8, and had the issue with the compiler not being chosen (I chose unspecified and GCC 10.3.1 arm-none-eabi) and both gave the error from the OP.

Bob

ParrSt, 2023-07-07:
Okay, I tried looking up how to set VS Code for Ninja CMake, and I am even more confused.  Is Ninja a compiler or a CMake tool?  I know how to program C and a little C++ but this compiling environment is really a beast!  Getting all the parts to play nice and now having to learn another language (CMake) that is confusing and seems redundant.  I so want to program the Pico but this learning curve is killing me.  

Any help getting it set up will be appreciated.

Bob

ParrSt, 2023-07-07:
Okay, I have been messing around with this for days now and getting nowhere.  I have googled till my fingers bled... okay maybe too much drama.  If I wanted to start over from scratch, short of re-installing a fresh windows 11 OS, what would I need to uninstall and or delete?  I am pretty sure this is due to my original configuration, which I uninstalled and deleted what I could find, leaving setting/config files or registry entries.

Anyone tell me what I would have to delete/uninstall for a fresh go at the new windows setup?

Bob

ParrSt, 2023-07-08:
Okay, I have found out how to clean uninstall VS Code and re-installed everything.  It mostly seems to be working but I am getting a message saying
`cannot open source file "pico/version.h" (dependency of "D:\RPPico\test\pico\stdlib.h")`
when following Chapter 8 of Getting Started.  I seems to compile fine.  Can't test since the last program did the same thing.  But Debug stops in bbot2_w25q080.S at line 99.  If I try to step out I get a message
`Could not step out: "finish" not meaningful in the outermost frame. (from exec-finish --thread 1)`
If I play, it works, but doesn't stop at breakpoints or at main().

I'm kinda lost now.

Bob

ParrSt, 2023-07-08:
Okay, I found out how to uninstall VS Code with settings and seem to have figured most of this out.  I did have to copy the pico-sdk-tools and pico-sdk folders to the folder I want to use for my projects.  Are there any other folders needed, or another way to reference these (and any other folders) without using a full path to each #include command?  I would rather not have to clutter my '\Program Files\Raspberry Pi\Pico SDK v1.5.1' folder with all my projects.

Bob

lurch, 2023-07-08:
I don't use Windows myself, so you'll have to wait for @ndabas to help you with that part, but responding to your "Is Ninja a compiler or a CMake tool?" question...

The way I understand things, CMake is effectively a "meta build tool", which can generate build-files for a variety of other build-tools; with Ninja, Make and NMake all being in this "build tool" category. The relevant build-tool then reads its build-file to understand how to invoke the necessary compilers and linkers etc. to turn your source files (`blah.c` and `blah.h`) into an RP2040-compatible executable file (`blah.elf`) and UF2 file (`blah.uf2`, which is the file you actually copy to your Pico).

If you're having trouble getting things working with the automated Pico Windows Installer (it sounds like "I did have to copy the pico-sdk-tools and pico-sdk folders to the folder I want to use for my projects" might be breaking some of the assumptions that the Installer makes?), you might find Section 9.2 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf useful?

ParrSt, 2023-07-08:
> If you're having trouble getting things working with the automated Pico Windows Installer (it sounds like "I did have to copy the pico-sdk-tools and pico-sdk folders to the folder I want to use for [...]
Yeah, I think if I understood all the parts of the compiling process better, I could tell them where the files were.  But I can code somewhat, but have mostly stayed away from compiling due to all the parts that have to all work just right to get a program compiled.  And since I so rarely do compiled, I have never learned the ins an outs.

I will check that section out as the better I can understand all the parts, the easier it will be to fix problems.

Bob

ndabas, 2023-07-08:
I believe the easiest thing to do will be to:
- Delete all of the stuff you copied/moved around (pico-sdk, pico-sdk-tools, the examples folders, etc.)
- Re-install with the installer. Don't worry about uninstalling first if this is the only version you've installed.
- Follow the instructions in the ReadMe that's included with the installer.
  - There are instructions on initial set up and also how to create a minimal project.
  - Make sure you only use the "Pico - ..." shortcuts to launch VS Code or command prompts to build your projects.

The installer does give you an option to choose both the install path and the path where examples are copied. The examples location can be changed on the last page of the installer. Just click browse and choose wherever you want those.

[image]

[image]

There are a bunch of PICO_xxx environment variables set up by the shortcuts which will ensure that the SDK can find the compilers and tools.

One of the things where I see is that you need to select either 'Pico ARM GCC" or "Unspecified" when prompted by VS Code (the CMake Tools extension really) to select a compiler/kit. This is in the ReadMe - [here's a copy of that with nicer formatting](https://github.com/raspberrypi/pico-setup-windows/blob/master/docs/tutorial.md).

Also as already pointed out by @lurch you cannot take a sub-folder of the examples and expect it to build as-is; there are instructions in the ReadMe on how to create a minimal project to start with.

ParrSt, 2023-07-08:
Thanks, I have it all pretty much working.  I know where the environmental variables are so I will check to see if I need to move any.  A re-install did not work until I deleted the 'code' folder from App data to get rid of old settings.  I think I am on the right path now.

Thank you both for your help.

closed 2023-07-08 by ParrSt