raspberrypi/pico-setup-windows #25: v1.5.1 does not create all registry keys anymore

paulober, 2023-07-04:
With the new v1.5.1 installer, no new key in "HKEY_LOCAL_MACHINE\SOFTWARE\WOW6432Node\Raspberry Pi" gets created anymore.

ndabas, 2023-07-04:
Right, so what is the issue with that really? We don't need this registry key as part of the functionality of the installer.

paulober, 2023-07-06:
Yes, that's right, not for the functionality for the installer but it's a good functionality for other projects and users as it's currently the simplest way to detect all installed Pico-SDKs (an their paths) if you don't want to scan the FS or for something like the uninstaller in the registry. 
Also it doesn't make the installer faster or less complex if this registry key is removed.

ndabas, 2023-07-06:
I'd say checking the uninstall keys is the recommended approach if you want to detect the presence of this installer. This is how tools like winget check for installed packages, and is also how a lot of other (first party and third party) software detects, for example, if VS Code is installed. You can check, for example, `HKEY_LOCAL_MACHINE\Software\Microsoft\Windows\CurrentVersion\Uninstall\Raspberry Pi Pico SDK v1.5.1`.

Removing something isn't always about speed, performance, or complexity. Engineering is about doing exactly what's needed, and nothing more.

ndabas, 2023-07-06:
> Does the uninstall uninstall the uninstall key?

It is customary for the uninstall to uninstall the uninstall keys at the end of the uninstall process.

So... yes. :)

ndabas, 2023-07-13:
I could do that, and probably will at some point -- however that is going to be a low priority because:
- People who already have SDK v1.5.0 installed are not going to come back and look for an updated installer. So you'll need to handle the case of the current v1.5.0 installer as it was released.
- People who are installing the SDK afresh have little reason to choose the v1.5.0 installer over the latest one.

My recommendation to detect the installed versions and their paths would be:
1. Find all keys that look like: `HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\CurrentVersion\Uninstall\Raspberry Pi Pico SDK vX.Y.Z`. Additionally, while user installers are not planned currently, you might want to check `HKEY_CURRENT_USER` as well, and both the 32/64-bit versions of HKLM, if you wanted to be thorough/complete.
2. Extract the version from the key name, not the DisplayName (as that might change if we decide to localize the installer at some point.)
3. Extract the path from UninstallString.

This will work for all current and future releases.

ndabas, 2024-01-25:
Closing as no further action is needed.

closed 2024-01-25 by ndabas