raspberrypi/pico-setup-windows #29: Problem finding path with new project

Sarahelma, 2023-08-29:
hi
I've followed your tutorials and it's working fine when I'm running the examples, when I've tried to create a new project using it this is what I'm getting

    PS C:\> copy ${env:PICO_SDK_PATH}\external\pico_sdk_import.cmake .
    copy : Cannot find path 'C:\external\pico_sdk_import.cmake' because it does not exist.
    At line:1 char:1
    + copy ${env:PICO_SDK_PATH}\external\pico_sdk_import.cmake .
    + ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (C:\external\pico_sdk_import.cmake:String) [Copy-Item], ItemNotFoundExce
    ption
    + FullyQualifiedErrorId : PathNotFound,Microsoft.PowerShell.Commands.CopyItemCommand

What am I doing wrong? Is it to do with setting the pico sdk path in environment variables?

ndabas, 2023-08-29:
Yes, looks like the `PICO_SDK_PATH` environment variable isn't set -- did you run this command from a _Pico - Developer PowerShell_ prompt?

Sarahelma, 2023-08-29:
Yeah it was because I opened PowerShell normally, it works now!
Thanks for your help

closed 2023-08-29 by Sarahelma