raspberrypi/pico-setup-windows #30: Installer Cant Get Pased The Choose Location

Cyberbulat, 2023-09-04:
[image]
As Shown In The Image It Dosent Let Me Get Passed This Screen Even If I Change The Location
[image]

ndabas, 2023-09-06:
The _Install_ button would be disabled in scenarios such as:
- Not enough free space on the selected drive: you need about 1.3 to 1.5 GB free for the current versions of the installer.
- Installation to the root of a drive (such as F: in your second screenshot) is not permitted. You need to select a folder on the drive to install to.

Cyberbulat, 2023-09-06:
Thank you, That some how crossed my mind! thanks!