raspberrypi/pico-setup-windows #32: Keeping the SDK repo updated as often as possible

Nippius, 2023-09-20:
Hi,

Just used this setup for the first time and was wondering if there is a correct way to update the installed SDK?

I like to keep the SDK updated as often as possible but the installer seems to create a shallow copy of SDK repo so I manually deleted the it and did a full clone to the same place so that I can `git pull` every so often.

Will this cause any issues (haven't personally found any so far) or is there a better way?

Thanks

ndabas, 2023-10-13:
The best way is to clone the SDK somewhere outside the paths managed/copied by this installer, and set `PICO_SDK_PATH` to that location when building your project.

The SDK repo included with the installer is shallow on purpose, as that is the released version which is part of the bundle that we have tested, and if you change anything -- it's no longer the same set of tools and code that we have validated.

Nippius, 2023-10-16:
Ok I see, thank you. I'll revert the changes to restore the shallow copy. I could configure `PICO_SDK_PATH` but I don't like to do that for all projects because I'm pretty sure I'll forget it often :/

Thanks once again!

closed 2023-10-16 by Nippius