raspberrypi/pico-setup-windows #34: Installer not working on Windows 11 on Arm64

Nabla128k, 2023-10-26:
Installer refuse to work on Arm64 Windows 11. Is there a reason for this?

ndabas, 2023-10-27:
Yes. We haven't validated that the toolchain (which is all built for x64) works correctly under emulation.

Nabla128k, 2023-12-04:
Toolchain itself works well if manually installed. But not the installer.

gbaman, 2024-04-21:
Just following up on this, any luck at removing the limitation for Arm64 version of windows?

ebadger, 2024-04-27:
+1 for a native ARM64 tool chain

ndabas, 2024-04-28:
This will be done in the next release of the installer -- which will coincide with the next release of the Pico SDK.

Note that there is no fully native ARM64 toolchain, as we include the ARM GCC and other tools from various vendors, and many do not have native ARM64 releases. We will include the tools that do have ARM64 releases, though.

closed 2024-08-15 by ghollingworth