raspberrypi/pico-setup-windows #36: Use of $env:COMSPEC may be invalid in some cases.

U-1F992, 2023-11-28:
In certain environments, this will be reproduced by running the installer as usual and launching Pico - Developer PowerShell.

Sorry for the Japanese UI, but it says "The expression after '&' in a pipeline element produced an object that was not valid. It must result in a command name, a script block, or a CommandInfo object."

[image]

This indicates that `$env:COMSPEC` is not defined. In fact, `echo $env:COMSPEC` did not print anything. Of course, I did not intentionally erase it... perhaps.

The disappearance seems to be a common occurrence, as evidenced by reports of similar problems occurring in other projects (other 3rd party applications supposedly erasing it? I have not investigated further).
ref: https://github.com/ScoopInstaller/Scoop/issues/3845

Anyway, `$env:COMSPEC` is just an environment variable, and since it can disappear or contain strange things, some people think it is wise to specify `cmd.exe` directly if we obviously know we will use it.
ref: https://github.com/ScoopInstaller/Scoop/pull/4692

ndabas, 2023-11-28:
Thanks for the bug report/debugging and resolution! I will fix this for the next release.

closed 2024-08-15 by ghollingworth

U-1F992, 2024-08-15:
@ghollingworth No modifications have been made since this issue was opened, and `$env:COMSPEC` is still being used. https://github.com/raspberrypi/pico-setup-windows/blob/e412aa4ca5ae329e73f65856eb5e55b57eb47306/packages/pico-setup-windows/pico-env.ps1#L6

Could you please explain why this issue was closed as completed?