raspberrypi/pico-setup-windows #37: No suitable code signing certificates found

chgenly, 2023-11-28:
Running setup on windows 10 I get the following failure.

[image]

ndabas, 2023-11-28:
[As noted in the ReadMe](https://github.com/raspberrypi/pico-setup-windows#pico-setup-for-windows), if you just want to install the tools, download the installer.

The build.ps1 file is for building the installer itself. If that is what you are trying to do, you can add the `-SkipSigning` argument.

chgenly, 2023-11-28:
Really closed

closed 2023-11-28 by chgenly

lurch, 2023-11-29:
@ndabas I wonder if moving the "Building" part of the Readme to a separate markdown file, might prevent other users from making a similar mistake?

ndabas, 2023-11-30:
@lurch Agreed, I think we can split that out to a separate file for developers.