raspberrypi/pico-setup-windows #38: GDB Error 1 in Visual Studio Code on Windows

progvault, 2023-12-22:
Clicking the bottom debug icon in Visual Studio Code produces an error: 
```
[proc] Executing command: "C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/gcc-arm-none-eabi/bin/arm-none-eabi-gdb.exe" --version
[proc] The command: "C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/gcc-arm-none-eabi/bin/arm-none-eabi-gdb.exe" --version exited with code: 1
[proc] Executing command: gdb --version
[proc] The command: gdb --version exited with code: 1
```

Are the forward slashes correct?

ndabas, 2023-12-22:
The forward slashes are correct, but you need to use the [configured profile](https://github.com/raspberrypi/pico-setup-windows/tree/master/packages/pico-examples/ide/vscode) for debugging, so the debug button on the bottom bar does not work. Our VS Code configuration hides that button, did you follow the steps in [the tutorial](https://github.com/raspberrypi/pico-setup-windows/blob/master/docs/tutorial.md) to set that up and start debugging?

progvault, 2023-12-22:
The VS Code configuration (i.e. .vscode folder) was supplied with the SDK and installed by pico-setup-windows-x64-standalone.exe.  

A zip file named pico-examples.zip contained the .vscode folder.

I should not that there was a step in the SDK installation process that failed.  I attempted to install the examples to the `C:\Program Files` folder instead of the C:\Users\  Documents directory.  The installer failed to extract the examples from github due to permissions error.

progvault, 2023-12-22:
I reinstalled.  Looks like the SDK installer does the same thing I did manually which was unzip pico-examples.zip to the documents directory.

Are you saying the debug icon should not be here?
[image]

progvault, 2023-12-23:
I've come to the conclusion that CMake is a mess.  Sure it works however debugging issues like this is a nightmare.  There is so much cache and crap left behind that pin-pointing the root cause of an issue is pretty much impossible.  Even resetting and clearing doesn't get rid of everything.

ndabas, 2023-12-24:
Yes, you're not supposed to use the debug button on the bottom status bar, the .vscode configuration that we provide is supposed to hide that for this reason. I'll see if I can replicate this behavior.

The recommended way to start debugging is documented in the [tutorial](https://github.com/raspberrypi/pico-setup-windows/blob/master/docs/tutorial.md) (which is included and installed as readme.txt if you use this installer.)

closed 2024-08-15 by ghollingworth