raspberrypi/pico-setup-windows #40: The system cannot find the file specified.

sdwood68, 2024-01-26:
On Windows 11 x86-64
After install finishes and when it starts to install the pico-examples into the def

```
The system cannot find the file specified.
Copying pico-examples...
New-Item : Could not find file 'Pico-v1.5.1'.
At
C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules\Microsoft.PowerShell.Archive\Microsoft.PowerShell.Archive.psm1:318
char:32
+ ... eatedItem = New-Item -Path $DestinationPath -ItemType Directory -Conf ...
+                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : WriteError: (C:\Users\Stuart\Documents\Pico-v1.5.1:String) [New-Item], FileNotFoundExcep
   tion
    + FullyQualifiedErrorId : CreateDirectoryIOError,Microsoft.PowerShell.Commands.NewItemCommand

C:\Program Files\Raspberry Pi\Pico SDK v1.5.1>
```

ndabas, 2024-01-28:
Looks like it couldn't create the `C:\Users\Stuart\Documents\Pico-v1.5.1` directory for some reason (The `The system cannot find the file specified.` is coming from a `mkdir` command.) Is there any reason you can think of, why this failed?

sdwood68, 2024-01-28:
Thanks for the reply.

I resolved the issue, by reinstalling Windows 11 and wiping the user
directory. I think the permission or ownerships were corrupted.

closed 2024-01-31 by ndabas