raspberrypi/pico-setup-windows #44: Embedded Systems Engineer

benji7480, 2024-06-12:
Embedded Systems Engineer

benji7480, 2024-06-12:
i just did the pico SDK set up on windows 10 and the build all option for the pico examples repository throws a configuration error that i am not able to solve. any help wud be gr8. pasting the error/problem below from the Pico visual studio window . 
Cmake seems to encounter the error in the CMakeLists.txt file line 12 in the pico-extras folder.

Slion, 2024-06-12:
You would need to provide a lot more info than that.

ndabas, 2024-06-13:
> pasting the error/problem below from the Pico visual studio window

Seems you forgot to do that. Please add more info and reopen the issue if needed.

closed 2024-06-13 by ndabas

benji7480, 2024-06-14:
> You would need to provide a lot more info than that.

Cmake encounters an error as below with the CMakeLists.txt file line no 12 in the pico extras folder :

line 12 begins with "    project(pico_extras C CXX)"

The actual Cmake error below when it encounters line 12 above :

CMake Error at C:/Users/bvu3kor/Documents/Pico-v1.5.1/pico-extras/CMakeLists.txt:12 (project):Running

 'nmake' '-?'

failed with:

 The system cannot find the file specified
-- Configuring incomplete, errors occurred!
See also "C:/Users/bvu3kor/Documents/Pico-v1.5.1/pico-examples/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/bvu3kor/Documents/Pico-v1.5.1/pico-examples/build/CMakeFiles/CMakeError.log".