raspberrypi/pico-setup-windows #45: Just flash ELF without debugging

Slion, 2024-06-13:
I'm trying to streamline my workflow by not having to mount the boot drive and copy the `uf2`.
I have not find a way to do that reliably from Visual Studio Code but it could be I just missed it.

Trying to do it from the [command line](https://github.com/raspberrypi/pico-setup-windows/blob/master/docs/tutorial.md#command-line-usage) seems to work but then I would need a one liner cause as it is I'm still faster mounting the drive and copying the file.

See also:
https://github.com/raspberrypi/debugprobe/issues/141

ndabas, 2024-06-13:
If you have a debug probe connected, you can use OpenOCD to just flash the ELF. There is a ["flash" task included](https://github.com/raspberrypi/pico-setup-windows/blob/master/packages/pico-examples/ide/vscode/tasks.json) in the VS Code configuration that we ship.

You could use that (it's a bit clunky in the VS Code UI) or the equivalent one-liner:

```sh
openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000; program {path/to/elf} verify reset exit"
```

closed 2024-06-13 by Slion

ndabas, 2024-06-13:
Great, glad that helped!

I think you might want to look into [launch configurations](https://code.visualstudio.com/docs/editor/debugging#_launch-configurations) -- we added that Flash task (and did not add the build task as a dependency) for the [C++ debugger task](https://github.com/raspberrypi/pico-setup-windows/blob/master/packages/pico-examples/ide/vscode/launch.json) that we have in the configuration (the last one in the file), because that task builds the code but does not flash it by default. You might be able to create a configuration which just does the build and flash, without lauching a debugger.

Slion, 2024-06-13:
To bind Flash that depends on Build to Ctrl+Shift+B I used that:
```
{
  "version": "2.0.0",
  "tasks": [
    {
      "label": "Flash",
      "type": "shell",
      "command": "openocd",
      "args": [
        "-f",
        "interface/cmsis-dap.cfg",
        "-f",
        "target/rp2040.cfg",
        "-c",
        "adapter speed 5000; program {${command:cmake.launchTargetPath}} verify reset exit"
      ],
      "problemMatcher": [],            
      "dependsOn": ["Build"],
      "group": {
        "kind": "build",
        "isDefault": true
      }
    },
    {
      "label": "Build",
      "type": "cmake",
      "command": "build",
      "problemMatcher": "$gcc",
    }
  ]
}
```

Just moved the following block from the Build task to the Flash task:
```
      "group": {
        "kind": "build",
        "isDefault": true
      }
```