raspberrypi/pico-setup-windows #47: Unable to uninstall Pico SDK

Intechgreater, 2024-07-13:
To uninstall, the instructions say to 
> Open the Apps and Features section in Windows Settings, then select Raspberry Pi Pico SDK <version>. Click the Uninstall button and follow the prompts.

There's nothing with the term Pico in my Installed Apps, nor start Menu, yet everything seemed to installed successfully and I've been utilizing the SDK in Arduino IDE projects.
I want to move/reinstall the SDK to a different directory that doesn't get backed up on OneDrive.

Windows 11 Pro
23H2

ndabas, 2024-07-14:
Do you think it's possible that you _didn't_ actually use this installer to install the SDK? We don't support and don't configure anything in the Arduino IDE. So if you're using it there you probably installed it via a different method.

What folders do you have which you believe were actually installed by this installer?

Intechgreater, 2024-07-20:
I followed this [Tom's Hardware article](https://www.tomshardware.com/how-to/program-raspberry-pi-pico-with-arduino-ide) that contains a link directly to [releases in your Github repo](https://github.com/ndabas/pico-setup-windows/releases) and used [pico-setup-windows-0.3.5-x64.exe](https://github.com/ndabas/pico-setup-windows/releases/download/v0.3.5/pico-setup-windows-0.3.5-x64.exe) since that seemed to be the correct one at the time. Later I looked around and noticed the Readme note on the main page saying that the project has moved to this Raspberry Pi repo which is why I ended up reporting the issue here. Having retraced my steps I realize I should have reported it on your repo instead, sorry. 

These were created by the installer (except Outlook?)
[image]

Should I just uninstall each manually and then reinstall via [pico-setup-windows-x64-standalone.exe](https://github.com/raspberrypi/pico-setup-windows/releases/latest/download/pico-setup-windows-x64-standalone.exe) from this repo?

ndabas, 2024-07-20:
Ah, good to know that that article has an outdated link now, I'll see if I can put a note on that release to prevent users doing this in the future.

Anyway, at the time of the release, the installer used to just install everything separately (as recommended in the Pico SDK docs), so you can uninstall those pieces if you want to, although there is no harm if you let them hang around. Then just delete the directory it copied to your Documents, as documented in the [ReadMe.txt](https://github.com/ndabas/pico-setup-windows/blob/v0.3.5/docs/ReadMe.txt) that would be present in there:
> Uninstall
> Just delete this directory. Any other software that you installed using Pico
setup for Windows will have its own uninstaller.

Either way, yes, you can install via the latest release in this repo.

closed 2024-08-15 by ghollingworth