raspberrypi/pico-setup-windows #53: Release for SDK 2.0.0

Manu3l0us, 2024-09-13:
I just took a Pico 2 out of the box and wanted to get things going. Unfortunately, there is not Windows installer release for SDK 2.0.0.

Manu3l0us, 2024-09-13:
Nevermind, just saw the note in the readme

closed 2024-09-13 by Manu3l0us

eriklundh, 2024-09-13:
I have used this installer to set up Pico-SDK for Pico development up until Pico-SDK 1.51.
I have now switched to https://github.com/raspberrypi/pico-vscode
One advantage of the pico-vscode plugin, and it's associated command line python utilities, is that it sets up your projects so that you don't have to start up with a special command file to get compilation to work. 
But I recommend that you learn of VSCode profiles, if you want to use the same VsCode install for any other purpose than Pico development, due to well-known issues with conflicting VsCode plugins.

Manu3l0us, 2024-09-13:
@eriklundh Thanks, I just got started with the VScode plugin. Mainly I'm interested in the Windows binaries of the tooling like OpenOCD. This worked out fine.